rednblkx/HomeKey-ESP32 #184: HomeKey not working on other user's devices

plaetzchen, 2025-04-06:
Hi,

Thank you for all the hard work on this project, it's really great to see that somebody takes this up. For hardware I am using the CASMo_NFC board. I am trying to set up 0.6 in my home, running 18.4.. It works fine with my device, added it, get a card in wallet, can tap to unlock from both of my devices (iPhone & Watch) and the state is reported through MQTT. Great!

So now I am trying to get it running with my partner's devices. They got the notification in Home to set it up and added it, took a moment but it's on the device and wallet. And now it just doesn't work. The card shows up on the device when tapping, there is a checkmark on the phone screen but there is not MQTT event triggered. I attached the complete debug log. 

Rebooting the device and removing it and adding it back to HomeKit did not help. My partner is a Home Admin.

Thanks,
plaetzchen

[debug_partner.txt](https://github.com/user-attachments/files/19620849/debug_partner.txt)

Damageplan1980, 2025-07-21:
Hi there,

same problem here. Any solution for this?

Thanks
Matthias

rednblkx, 2025-08-15:
Hi!

Sorry for the extremely delayed reply.

Please try the latest CI build and let me know

https://github.com/rednblkx/HomeKey-ESP32/actions/runs/16989491332

plaetzchen, 2025-09-03:
Hey,

So I finally found the board, installed the merged binary but can not get it to add to HomeKit. It runs into a timeout

plaetzchen, 2025-09-03:
So kind of funny but it now works under the latest iOS 18 version.

rednblkx, 2025-09-14:
Glad it works now, the timeout issue is a bug from the Home app from my testing but there could also be something on my side I haven’t noticed, usually helps if you reconnect to the WiFi so it can clear any cache. I’ll close this issue, let me know if you have any other issues.

closed 2025-09-14 by rednblkx