rednblkx/HomeKey-ESP32 #196: Try Again Homekey error

technogrady, 2025-08-12:
Me and My roommate get this error on our phones and watches pretty often but not all the time. Not sure why it's happening. [image]
thank you all!

hetrodoo, 2025-08-12:
Hi, I'm not a developer on this project but I've been using this flawlessly for the past month, either with my phone or watch.
This only happens to me when I interrupt the communication by removing the phone or watch from the sensor quickly or with a bad PN532 that I have.

What hardware are you using (esp32, c3, s3)? Do you have any logs at the time of the error? [Is you PN532 a clone board](https://github.com/rednblkx/HomeKey-ESP32/wiki/Prerequisites#1-hardware)?

technogrady, 2025-08-12:
[[This one](https://a.co/d/5IUTzEb)](https://a.co/d/5IUTzEb) I bought for this one and it doesn't have the blue components but components in same spot of similar size. Purchased a 2 pack to do it to another door and those seem to have blue components.