rednblkx/HomeKey-ESP32 #198: A stack overflow in task alt_action_task has been detected

a2852a, 2025-08-16:
Hi, there.

I'm trying to setup HomeKey (v0.6 from Feb 19) on T-ETH-Lite ESP32-S3 LilyGO WiFi W5500 (manufacturer code 5905523320282).

When I assign pin to `alt key` I receive this error:
```
I (2737) LockManagement: LockManagement:311 > Configuring LockManagement
I (2743) LockMechanism: LockMechanism:554 > Configuring LockMechanism
I (2751) NFCAccess: NFCAccess:608 > Configuring NFCAccess
I (2757) MAIN: alt_action_task:374 > Starting Alt Action button task

***ERROR*** A stack overflow in task alt_action_task has been detected.

Backtrace: 0x403756ce:0x3fcb5060 0x4037e595:0x3fcb5080 0x4037f1b6:0x3fcb50a0 0x403806a5:0x3fcb5120 0x4037f2e8:0x3fcb5150 0x4037f2de:0x0000000c |<-CORRUPTED

ELF file SHA256: 18b433faf

Rebooting...

```

Everything else works fine.

Replicated this on fresh install and both with connected other stuff on gpio's and without. Also tried to assign different pins - no difference.

May I ask for help with this issue? 🙏

a2852a, 2025-09-19:
@rednblkx Hey there, any update related to this issue? 🤞

rednblkx, 2025-09-20:
Yes, it is fixed but it’s part of a much larger update I’m working on, you’re free to try the latest CI build but note that a lot has changed and it is still being actively worked on.

closed 2025-11-23 by rednblkx